circuitdiagram/circuitdiagram #1: Circuit Diagram has stopped working

mrpengilley, 2015-08-08:
When I try and open the new stable version of Circuit Diagram I receive the following message:

"Circuit Diagram has stopped working
A problem caused the program to stop working correctly. Windows will close the program and notify you if a solution is available."

Do you know what is happening here?

The Beta version was working fine no problems.

SamuelFisher, 2015-08-08:
Could you post a copy of the error log to help find the problem? You can normally find it at:

`C:\Users\Username\AppData\Roaming\Circuit Diagram\ErrorLog.txt`

Thanks

julienboriasse, 2015-09-03:
Here is mine:

System.IO.FileLoadException: Could not load file or assembly 'cdlibrary, Version=1.0.0.0, Culture=neutral, PublicKeyToken=18fa230cbe445e9d' or one of its dependencies. The located assembly's manifest definition does not match the assembly reference. (Exception from HRESULT: 0x80131040)
File name: 'cdlibrary, Version=1.0.0.0, Culture=neutral, PublicKeyToken=18fa230cbe445e9d'
   at CircuitDiagram.App.Application_Startup(Object sender, StartupEventArgs e)
   at System.Windows.Application.<.ctor>b__1(Object unused)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at MS.Internal.Threading.ExceptionFilterHelper.TryCatchWhen(Object source, Delegate method, Object args, Int32 numArgs, Delegate catchHandler)

=== Pre-bind state information ===
LOG: DisplayName = cdlibrary, Version=1.0.0.0, Culture=neutral, PublicKeyToken=18fa230cbe445e9d
 (Fully-specified)
LOG: Appbase = file:///C:/Program Files/Circuit Diagram/
LOG: Initial PrivatePath = NULL
Calling assembly : Circuit Diagram, Version=3.1.0.3, Culture=neutral, PublicKeyToken=null.
===
LOG: This bind starts in default load context.
LOG: No application configuration file found.
LOG: Using host configuration file: 
LOG: Using machine configuration file from C:\Windows\Microsoft.NET\Framework64\v4.0.30319\config\machine.config.
LOG: Post-policy reference: cdlibrary, Version=1.0.0.0, Culture=neutral, PublicKeyToken=18fa230cbe445e9d
LOG: Attempting download of new URL file:///C:/Program Files/Circuit Diagram/cdlibrary.DLL.
WRN: Comparing the assembly name resulted in the mismatch: Major Version
ERR: Failed to complete setup of assembly (hr = 0x80131040). Probing terminated.

adonios77, 2026-10-02:
So, whats happening with this error ?
I'm getting the same error message either for 3 & 3.1 versions:
**"Circuit Diagram has stopped working
A problem caused the program to stop working correctly. Windows will close the program and notify you if a solution is available."**

My **`ErrorLog.txt`** from `"C:\Users\%UserName%\AppData\Roaming\Circuit Diagram\"`:
``` 
System.Windows.Markup.XamlParseException: The invocation of the constructor on type 'CircuitDiagram.MainWindow' that matches the specified binding constraints threw an exception. ---> System.Runtime.InteropServices.COMException: The Server service is not started.

   at System.DirectoryServices.DirectoryEntry.RefreshCache()
   at System.DirectoryServices.AccountManagement.PrincipalContext.DoMachineInit()
   at System.DirectoryServices.AccountManagement.PrincipalContext.Initialize()
   at System.DirectoryServices.AccountManagement.PrincipalContext.get_QueryCtx()
   at System.DirectoryServices.AccountManagement.Principal.FindByIdentityWithTypeHelper(PrincipalContext context, Type principalType, Nullable`1 identityType, String identityValue, DateTime refDate)
   at System.DirectoryServices.AccountManagement.Principal.FindByIdentityWithType(PrincipalContext context, Type principalType, IdentityType identityType, String identityValue)
   at System.DirectoryServices.AccountManagement.UserPrincipal.FindByIdentity(PrincipalContext context, IdentityType identityType, String identityValue)
   at System.DirectoryServices.AccountManagement.UserPrincipal.get_Current()
   at CircuitDiagram.MainWindow.ApplySettings()
   at CircuitDiagram.MainWindow..ctor()
   --- End of inner exception stack trace ---
   at System.Windows.Markup.XamlReader.RewrapException(Exception e, IXamlLineInfo lineInfo, Uri baseUri)
   at System.Windows.Markup.WpfXamlLoader.Load(XamlReader xamlReader, IXamlObjectWriterFactory writerFactory, Boolean skipJournaledProperties, Object rootObject, XamlObjectWriterSettings settings, Uri baseUri)
   at System.Windows.Markup.WpfXamlLoader.LoadBaml(XamlReader xamlReader, Boolean skipJournaledProperties, Object rootObject, XamlAccessLevel accessLevel, Uri baseUri)
   at System.Windows.Markup.XamlReader.LoadBaml(Stream stream, ParserContext parserContext, Object parent, Boolean closeStream)
   at System.Windows.Application.LoadBamlStreamWithSyncInfo(Stream stream, ParserContext pc)
   at System.Windows.Application.DoStartup()
   at System.Windows.Application.<.ctor>b__1_0(Object unused)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
```
(Also, none of the v4.x-pre.x versions works for my Windows 7 x64 OS)