circuitdiagram/circuitdiagram #6: Circuit Diagram stopped working

harjunmaa, 2016-07-20:
I boot my PC (Windows 8), and without doing anything else, I start Circuit Diagram and open my file. Then, without doing anything with it, I Save it. Comes up error message "Circuit Diagram has stopped working..."

Here is the error log:

System.IO.IOException: The process cannot access the file 'C:\Users\Hannu\Documents\Valoa\OD600\OD600circuit1.cddx' because it is being used by another process.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.FileStream.Init(String path, FileMode mode, FileAccess access, Int32 rights, Boolean useRights, FileShare share, Int32 bufferSize, FileOptions options, SECURITY_ATTRIBUTES secAttrs, String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, String msgPath, Boolean bFromProxy)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share)
   at CircuitDiagram.MainWindow.SaveDocument(Boolean& saved)
   at CircuitDiagram.MainWindow.CommandSave_Executed(Object sender, ExecutedRoutedEventArgs e)
   at System.Windows.Input.CommandBinding.OnExecuted(Object sender, ExecutedRoutedEventArgs e)
   at System.Windows.Input.CommandManager.ExecuteCommandBinding(Object sender, ExecutedRoutedEventArgs e, CommandBinding commandBinding)
   at System.Windows.Input.CommandManager.FindCommandBinding(CommandBindingCollection commandBindings, Object sender, RoutedEventArgs e, ICommand command, Boolean execute)
   at System.Windows.Input.CommandManager.FindCommandBinding(Object sender, RoutedEventArgs e, ICommand command, Boolean execute)
   at System.Windows.Input.CommandManager.OnExecuted(Object sender, ExecutedRoutedEventArgs e)
   at System.Windows.RoutedEventArgs.InvokeHandler(Delegate handler, Object target)
   at System.Windows.EventRoute.InvokeHandlersImpl(Object source, RoutedEventArgs args, Boolean reRaised)
   at System.Windows.UIElement.RaiseEventImpl(DependencyObject sender, RoutedEventArgs args)
   at System.Windows.Input.RoutedCommand.ExecuteImpl(Object parameter, IInputElement target, Boolean userInitiated)
   at System.Windows.Controls.MenuItem.InvokeClickAfterRender(Object arg)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at MS.Internal.Threading.ExceptionFilterHelper.TryCatchWhen(Object source, Delegate method, Object args, Int32 numArgs, Delegate catchHandler)

This cannot be right, because there IS NO OTHER process going that understands cddx files. So what could be wrong? Circuit Diagram downloaded 7/14/2016, has worked fine for six days.

SamuelFisher, 2016-07-25:
Would you be able to check which process is using the file when Circuit Diagram tries to open it?

You can use [Process Explorer](https://technet.microsoft.com/en-gb/sysinternals/bb896653) to find out.

Thanks

harjunmaa, 2016-07-25:
No process was using it, really. But the next day, the problem had disappeared without explanation, and hasn't come back - working fine.Hannu-----Original Message-----
From: Samuel Fisher 
Sent: Jul 25, 2016 4:27 PM
To: circuitdiagram/circuitdiagram 
Cc: harjunmaa , Author 
Subject: Re: [circuitdiagram/circuitdiagram] Circuit Diagram stopped working (#6)

Would you be able to check which process is using the file when Circuit Diagram tries to open it?

You can use Process Explorer to find out.

Thanks

  
  

Hannu Harjunmaa, PhD
President
Valoa Technologies, Inc.
14 Damon Road
Holden, MA 01520

http://igg.me/at/No-more-fingerpricks

SamuelFisher, 2016-07-25:
Glad to hear it's working now.

closed 2016-07-25 by SamuelFisher