circuitdiagram/circuitdiagram #13: how to fill color of <rect /> with colour likr red , green

naveenuniyal, 2018-03-19:
Hey all ,

When drawing new component in component editor , i unable to fill the rectangle with color .

Is it possible to fill the < rect /> with other colur  like red , green ?

<path start="_Middle-20x-6y" data="l 6,6 l -6,6" thickness="1"  fill="true" />

this only fill black colour .

Thanks you in advance !

SamuelFisher, 2018-03-19:
I'm afraid it's not possible to fill a `<rect />` with colour. All of the component drawing commands draw in black only.

naveenuniyal, 2018-03-20:
@SamuelFisher : thanks for the prompt reply .

Just want to know , Is there any other way to add colour ? may be need any update in component editor ? , (because why i am saying  , as i read SVG tutorial ,over there its possible to draw shapes filled with different colour , so just wondering )

or any property we can add directly in circuit diagram code ? I mean i c# code ?

Thanks !

SamuelFisher, 2018-03-20:
The syntax for drawing in components is a subset of what is supported by SVG just to make it easier for people to make components who have worked with SVG before. Adding colour would require changes to the C# code as you suggest.

I'm not sure adding colour would be a good fit for circuit diagrams that are normally black and white. What is the use case you have in mind for this?

closed 2018-05-13 by SamuelFisher

naveenuniyal, 2018-05-14:
Thanks @SamuelFisher for the help :) ,
Just to add on , color should be added in diagram , because their are institution or industry , where they coded the circuit diagram with colors .

Closed #13<https://github.com/circuitdiagram/circuitdiagram/issues/13>.