circuitdiagram/circuitdiagram #17: Installing Visual Studio Code plugin fails

rlneumiller, 2019-06-14:
6/14/2019 8:02:41 AM - Microsoft VSIX Installer
6/14/2019 8:02:41 AM - -------------------------------------------
6/14/2019 8:02:41 AM - vsixinstaller.exe version:
6/14/2019 8:02:41 AM - 15.8.3252
6/14/2019 8:02:41 AM - -------------------------------------------
6/14/2019 8:02:41 AM - Command line parameters:
6/14/2019 8:02:41 AM - C:\Program Files (x86)\Microsoft Visual Studio\Installer\resources\app\ServiceHub\Services\Microsoft.VisualStudio.Setup.Service\VSIXInstaller.exe,D:\downloads\circuitdiagram-vscode-0.3.0-dev.vsix
6/14/2019 8:02:41 AM - -------------------------------------------
6/14/2019 8:02:41 AM - Microsoft VSIX Installer
6/14/2019 8:02:41 AM - -------------------------------------------
6/14/2019 8:02:42 AM - Initializing Install...
6/14/2019 8:02:42 AM - Microsoft.VisualStudio.ExtensionManager.InvalidExtensionManifestException: Element 'Identity' has invalid value for attribute 'Version' ---> System.FormatException: Input string was not in a correct format.
   at System.Version.VersionResult.SetFailure(ParseFailureKind failure, String argument)
   at System.Version.TryParseComponent(String component, String componentName, VersionResult& result, Int32& parsedComponent)
   at System.Version.TryParseVersion(String version, VersionResult& result)
   at System.Version.Parse(String input)
   at System.Version..ctor(String version)
   at Microsoft.VisualStudio.ExtensionManager.ExtensionImpl.<>c.<ExtractManifestData>b__67_1(String s)
   at Microsoft.VisualStudio.ExtensionManager.ExtensionImpl.VerifyStringValue[T](String value, ErrorInfo info, Func`2 selector, Boolean isRequired, Int32 maxLength, T defaultValue)
   --- End of inner exception stack trace ---
   at Microsoft.VisualStudio.ExtensionManager.ExtensionImpl.VerifyStringValue[T](String value, ErrorInfo info, Func`2 selector, Boolean isRequired, Int32 maxLength, T defaultValue)
   at Microsoft.VisualStudio.ExtensionManager.ExtensionImpl.ExtractManifestData(PackageManifest manifest)
   at Microsoft.VisualStudio.ExtensionManager.ExtensionImpl.ReadPackageManifest(XmlReader reader)
   at Microsoft.VisualStudio.ExtensionManager.ExtensionImpl.ReadManifest(Stream manifestStream)
   at Microsoft.VisualStudio.ExtensionManager.InstallableExtensionImpl.ReadVSIXManifestFromPackage(Stream stream, CultureInfo preferredCulture)
   at Microsoft.VisualStudio.ExtensionManager.InstallableExtensionImpl..ctor(String path, CultureInfo preferredCulture)
   at Microsoft.VisualStudio.ExtensionManager.ExtensionEngineImpl.CreateInstallableExtension(String extensionPath, CultureInfo preferredCulture)
   at Microsoft.VisualStudio.ExtensionManager.ExtensionEngineImpl.CreateInstallableExtension(String extensionPath)
   at VSIXInstaller.ExtensionService.GetInstallableData(String vsixPath, String extensionPackParentName, Boolean isRepairSupported, IStateData stateData, IEnumerable`1& skuData)
   at VSIXInstaller.ExtensionPackService.IsExtensionPack(IStateData stateData, Boolean isRepairSupported)
   at VSIXInstaller.ExtensionPackService.ExpandExtensionPackToInstall(IStateData stateData, Boolean isRepairSupported)
   at VSIXInstaller.App.Initialize(Boolean isRepairSupported)
   at VSIXInstaller.App.Initialize()
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()

SamuelFisher, 2019-06-14:
Are you trying to install it from within VS Code or by double-clicking on the VSIX file?

Please try installing it from within VS Code using the `Install from VSIX...` option - see here for more details on how to install:

https://www.circuit-diagram.org/news/2018/05/visual-studio-code-extension

Thanks

closed 2019-06-17 by SamuelFisher

rlneumiller, 2019-06-18:
I'm sure that was it.  Might be worthwhile to mention that in the readme for other VSCode noobs such as myself.

rlneumiller, 2019-06-18:
How embarrassing!  I totally missed that somehow.  Apologies for the erroneous bug report.