circuitdiagram/circuitdiagram #19: SVG exports seem to misplace some stuff / not save everything / scale font weirdly

JambonSama, 2019-08-05:
My circuit diagram :
[image]

When I insert the svg in a pdf file : 
[image]

- misplacing : the 'M' in the motor component is not centered anymore
- not saving everything : every unicode character that isn't ASCII got clipped away
- scaling things weirdly : the font size is independant of the scale of my insertion of the svg in the pdf

I'm opening an issue because in fine it's an issue, but for now I have a work around anyway. 

(It's just a "letting you now that...".)

SamuelFisher, 2019-08-07:
I've tried this with the following example circuit and the exported SVG displays correctly in Google Chrome (including unicode characters and text scaling when zooming in/out):

**Editor:**

[image]

**SVG export displayed in Chrome:**

[image]

To help diagnose this issue please could you provide the following information:

- When you export the circuit as an SVG, does it display correctly if you try opening the SVG in Chrome?
- How are you inserting the SVG into a PDF file? E.g. software you are using to do this etc.

JambonSama, 2019-08-08:
I used those two files to test things out : 

- https://crcit.net/c/1a0327ae186646269b2ad47116c1043c without non-ascii character
- https://crcit.net/c/deead3c09246450c9103afb44d74a640 with non-ascii characters

Displaying in a browser works fine. I don't have Chrome, but I opened both downloaded svg files with Firefox, and there is not a single problem (every character appears, appears where it should, and scales correctly).

What I'm trying to attempt : drawing a circuit diagram properly, and inserting it as a figure in a pdf using LaTeX and the LaTeX package `svg` (https://ctan.org/pkg/svg?lang=en). Concerning the issues I reported : 
- the non-ascii character issue is on me : LaTeX actually issued an error message, but since a document was outputed, and there were other problems with the svg, I didn't actually bother to notice there was an error message, that's on me, sorry.
- the scaling issue and the displacing characters issue are reproducible, steps below. I have no idea where in the process it doesn't work :( 

Steps for a minimal working example of the two issues (scaling and displacement) : 

1. Make sure Inkscape is installed and accessible in the PATH (or whatever you have the is the equivalent of the PATH on your setup.
2. Download the non-ascii character document (link above) as an svg file with the name `circuit.svg`
3. In the download folder of the svg file, create a .tex file with the following code : 

    \documentclass[]{article}
    \usepackage{svg}

    \begin{document}
		
        \begin{figure}
    	    \includesvg[width=\textwidth]{circuit2}
        \end{figure}

    	\begin{figure}
	    	\includesvg[width=0.1\textwidth]{circuit2}
    	\end{figure}

    \end{document}

4. Compile with the command `--shell-escape`

I get that : 
[image]

Now, I know LaTeX can be a pain in the ass. If you're actually committed to trying to fix this hot mess, but have never used LaTeX / the `svg` package, that package can be tricky : 

- It doesn't work well on Windows (though it does work with some effort, see that posts that explains it much better than me : https://tex.stackexchange.com/a/74693/141925 (I linked the specific answer that explains how to do it on Windows)
- I use a Manjaro Linux, have Inkscape installed, never worried about ghostscript or imagemagick
- I don't know anything about macs.

There. I don't think I can provide more details, but of course, don't hesitate to ask.

closed 2019-08-08 by JambonSama