circuitdiagram/circuitdiagram #27: display error on components options when importing

papilinou, 2020-11-09:
when importing from PC a cddx project, inductor value is set to show even when set to off when exporting,and toggle switch show closed even when set to open at export,external input same,set reverse when re-importing,but if exporting the project to png ,settings are ok on png export.
I think that when components are reloading in display, you don't take care to components options like "display value" "reversed" and so on but they still are in cddx file.

SamuelFisher, 2020-11-09:
The problem may be that you are using different versions of these components in the Desktop version compared to the latest versions used in the Web Editor.

You can try replacing them with the latest versions that are available for download from here: https://www.circuit-diagram.org/components - however please be aware that these versions will not display icons in the toolbar in Circuit Diagram Desktop.

The installed versions to be replaced can normally be found under `C:\Program Files\Circuit Diagram\ext`.

papilinou, 2020-11-10:
i'am speaking about web editor not desktop

papilinou, 2020-11-10:
same with component "multiplexer"  which use Property type "bool"

SamuelFisher, 2020-11-10:
Ok, thank you for explaining this in more detail. The issue has now been fixed.

closed 2020-11-10 by SamuelFisher