circuitdiagram/circuitdiagram #29: "condition match/notmatch text"return always true

papilinou, 2020-11-13:
"condition match/notmatch text"return always true in the joined example:
"Text " value default =bar
conditions="$Text==foo" return true
conditions="$Text==doe" return true
[image]

SamuelFisher, 2020-11-15:
Please could you change the screenshot to be a text snippet like below:

```xml
<?xml version="1.0" encoding="utf-8"?>
<component ...
```

and attach screenshots to demonstrate what is rendered compared to what you are expecting?

SamuelFisher, 2020-11-17:
The problem is that you have `<component version="1.0" ...` which is an old version of the components XML format. In version 1.0, _conditions_ had a different syntax which is why the example you posted doesn't work correctly. Please could you try changing line 2 to:

```xml
<component version="1.4" ...
```

That should fix the issue. Please let me know how you get on with this.

papilinou, 2020-11-19:
Hi, changed to 1.4,eveything is fine, conditions are ok.
Is there a location to get more infos like those requirements when building a component ?
and could you have a look to CD components issue#112 about title block ?